sinara-hw/Kasli-DIOT_x86_board #2: Critical Mechanical Issues

gravelpitman, 2025-03-11:
Hi, 
I have been going through this pcb and found several errors including critical ones:

**1. Incorrect PCB shape. It won't be possible to put it in the DIOT crate.**

[image]
The PCB is too big in the Y-axis. It will collide with the rails in the crate.
Maximum Y-axis dimmension is 100mm. 

[image]
Orange edge should be moved 2mm down (Y-)

**2. Inadequate connector combination between Kasli and SBC PCB.** 

[image]
This combination of connector does not provide enough Z axis clearance (current combination provide from 6,09mm to 7,31mm)
We need **at least** 7,2mm clearance.
Combination of LSS-140-03-L-DV-A and LSS-140-01-L-DV-A provides from 7,09mm to 8,31mm (https://suddendocs.samtec.com/prints/lss%20mated%20views-mkt.pdf)
**3. Add SMT threaded spacer on the bottom of SBC PCB**

[image]
Add 9774075151R SMT threaded standoffs

[image]
This standoffs provide 7.5mm clearance between the PCBs.
**Note that these sleeves are for M2.5 threads, so Kasli DIOT will need changes**. Only M2,5 standoffs have 0,5mm increments.

**4. Connectors are too close to the edge and they are to high**

[image]
These connectors may colide with the rails. Especially bottom 4pin connector.

[image]
When you add the corresponding connector for the cable you are left with only ~1.4mm clearance for the wires. Measured to the bottom surface of the next module and to the rail. 

**5. Incorrect standoffs under SOM**
[image]
Standoffs are to low. Based on  PCIE MXM GEN 3.0 -
10151114-001TLF datasheet (https://cdn.amphenol-cs.com/media/wysiwyg/files/drawing/10151114.pdf), , you can select the appropriate length of standoffs.
[image]
9774050151R standoff should be used.

kaolpr, 2025-04-27:
@gravelpitman Can you please verify that CSP heat sink (see [datasheet](https://www.congatec.com/fileadmin/user_upload/Documents/Manual/SA80.pdf)) will fit?

gravelpitman, 2025-05-05:
@kaolpr  We have around 11 mm above the top of the CSP, so it will fit.