sinara-hw/Kasli-DIOT_x86_board #4: Heat managment problem

kaolpr, 2025-05-14:
Well, we (@gravelpitman and me) have messed up. CSP clearly does not fit. 

[image]

However, the problem is that even when HSP variant is used (flat heatspreader) we have only < 3 mm left. 

[image]

I will give HSP a shot to see how bad it is, but honestly I don't think this will be a viable solution.

However, it looks that we could fit a CSP if the SoM was placed directly on Kasli DIOT. Still clearance will be ~1mm. It would be VERY invasive (rather for Kasli DIOT v2). We would have to move SMARC connector just after SFP cages. Move current B2B connector in between FTDI and DDMTD oscillators and redesign x86 board to have only the part with connectors and SSD.

Don't even know if this is really possible regarding layout constraints.

gkasprow, 2025-05-14:
it looks like we have t ogo for custom heat spreader, common for FPGA and SOM

kaolpr, 2025-05-14:
And what do you think of moving SoM to Kasli DIOT?

If possible, it would let us use COTS heatsinks for SoM and will not limit us to a specific SoM.

gkasprow, 2025-05-14:
We have a minimal number of use cases. Not worth redesigning the Kasli DIOT board.
The custom heatsink is easy and not that expensive.